puku0x/cvdrone #22: How to run an example.

anfedres, 2015-05-12:
Greetings. 

Great job by the way. 

I'm little bit new at this. I just saw your library and it seems excelent. 

I have already run the make file in /cvdrone-master/build/linux. The output was test.a. However. I don't know what to do next. Can you give me a hint how I can compile and run one of the examples and how I could create my one program? Which are the linkers that I have to use? Thanks.

anfedres, 2015-05-12:
Nevermind! I did it. The make file was the answer. Just change for the example you want to compile.

closed 2015-06-19 by anfedres

MisterTellini, 2016-01-15:
Good morning everyone!

Going back to this question, as I'm also newbie with this and I'm stuck in the same issue.

In my case (using W8/10) is it needed to modify the 'makefile' in (...)\cvdroneEDIT\build\mingw to run the samples? In that case, do I have to replace, for instance, the "PROGRAM = " statement (appearing 'Test.exe') just for 'kalman_tracking.cpp'?

This is what I've done until now, unsuccessfully.

Thank you for your kind support!

valizada, 2016-01-15:
Hello,

"test.exe" is just a name given to the result program. So, it can be anything. What you need to modify is the line above "../../src/main.o", just change the path to the one you want to run. In case that, you want to run sample_kalman_tracking.cpp, you will need to replace "../../src/main.o" to "../../samples/kalman_tracking.o"

Regards,
Fiz

Zepolrotia92, 2016-01-15:
Hi Fiz,

Many thanks for your fast response! But it does not seem to work out... It compiles the 'main.o' file by default again.

And is this even without having declared in 'Source files'?

(As you may observed, I'm dull at Visual Studio...)

Regards,
Aitor