puku0x/cvdrone #25: Original Camera Image

venkateshbs, 2015-06-08:
Hi,
 I want to know if we could get the original camera image rather than the scaled image.
Thanks
Venkatesh B.S.

puku0x, 2015-06-14:
Hi,

You mean you want to get a 720p image from your AR.Drone 2.0 ? 
If so, you should comment out line 105 and uncomment line 107 in [command.cpp](https://github.com/puku0x/cvdrone/blob/master/src/ardrone/command.cpp).

Thank you
puku0x

venkateshbs, 2015-06-15:
Hi,
  I am referring to  

sws_scale(pConvertCtx, (const uint8_t* const*)pFrame->data, pFrame->linesize, 0, pCodecCtx->height, pFrameBGR->data, pFrameBGR->linesize);

in video.cpp

I want to process the original image rather than a scaled version.
For example when I switch to bottom camera, the image should be smaller.
Is that possible ? or will it involve tricky implementation.

Thanks
Venkatesh B.S.

puku0x, 2015-06-20:
Hi

I guess you are using AR.Drone 1.0.

```c++
// src\ardrone\video.cpp
ARDRONE_IMAGE ARDrone::getImage(void)
{
    // There is no image
    if (!img) return ARDRONE_IMAGE(NULL);

    // Enable mutex lock
    if (mutexVideo) pthread_mutex_lock(mutexVideo);

    // AR.Drone 2.0
    if (version.major == ARDRONE_VERSION_2) {
        // Copy current frame to an IplImage
        memcpy(img->imageData, pFrameBGR->data[0], pCodecCtx->width * ((pCodecCtx->height == 368) ? 360 : pCodecCtx->height) * sizeof(uint8_t) * 3);
    }
    // AR.Drone 1.0
    else {
        // If the sizes of the buffer and the IplImage are differnt
        if (pCodecCtx->width != img->width || pCodecCtx->height != img->height) {
            // Re-allocate image (not tested yet !)
            cvReleaseImage(&img);
            img = cvCreateImage(cvSize(pCodecCtx->width, pCodecCtx->height), IPL_DEPTH_8U, 3);
            img->imageData = (char*)bufferBGR;

            // Resize the image to 320x240
            //IplImage *small_img = cvCreateImageHeader(cvSize(pCodecCtx->width, pCodecCtx->height), IPL_DEPTH_8U, 3);
            //small_img->imageData = (char*)bufferBGR;
            //cvResize(small_img, img, CV_INTER_CUBIC);
            //cvReleaseImageHeader(&small_img);
        }
        // For 320x240 image, just copy it
        else memcpy(img->imageData, bufferBGR, pCodecCtx->width * pCodecCtx->height * sizeof(uint8_t) * 3);
    }
    
    // The latest image has been read, so change newImage accordingly
    newImage = false;

    // Disable mutex lock
    if (mutexVideo) pthread_mutex_unlock(mutexVideo);

    return ARDRONE_IMAGE(img);
}
```

If you are using AR.Drone 2.0, it is impossible, just forget about it :p

Regards,
puku0x

venkateshbs, 2015-08-10:
Hi,
 Sorry for late response. I am using AR. Drone 2.0.
Thanks for the code. It was very useful for me.
Any plans for Bebop ?
Thanks
Venkatesh B.S

puku0x, 2015-10-18:
Yeah !  Although it is expensive and I don't have it yet, I'm interested in it !!

closed 2015-10-18 by puku0x