puku0x/cvdrone #26: Yaw

anfedres, 2015-06-19:
Hi. I don't know if you have notice that the motion Yaw does not have the same speed turning to the left than turning to the right. 

I trying to do a controller and I'm having problems sending the commands. Thanks.

puku0x, 2015-06-20:
Hi

Thank you for your report.

As you can see in [command.cpp](https://github.com/puku0x/cvdrone/blob/master/src/ardrone/command.cpp), the current implementation of sending speed commands has no feedback controller.

To make the rotational speed the same in right and left, you should implement a feedback controller in yaw velocity by obtaining rotational speed or using a marker.

Regards,
puku0x

anfedres, 2015-06-20:
Thanks puku0x for your answer. In fact I did one. 

I tried sending a constant speed to the controller by pressing the right and left keys. However, if you check yourself. the speed of rotating to the left, is not the same speed than rotating to the right. So, the controller won't work if you send, let's say a speed of 0.65 will be a speed of rotating to the right, but sending -0.65 is not actually -0.65 rotating to the left. it could be -0.40. You know what I mean? 

I also followed the controller that you have in one of the examples. But it doesn't work due to this bug. Please, give it a try. All the motion controls work fine. The only motion control that I'm having troubles is Yaw. 

May I ask how did you found the quaternion transformation? I have seen that you found this experimentally. Can I ask which one was the process that you have followed to find those? Have you checked other libraries to see how did they have found this values? . I'm referring to this part. 

 float v[4] = {-0.2f * (float)vy, -0.2f * (float)vx, 1.0f * (float)vz, -0.5f * (float)vr};

The last but not least, congratulations for your great work. The library is very useful. I'm just having problem with the Yaw. 

anfedres.

puku0x, 2015-06-20:
Thank you for providing details.

Most of the other projects, the unit of command speeds was [%] and I thought it was not intuitive.

The magic values in ARDrone::move3d() are the coefficients which convert from [%] to [m/s] and [rad/s], and these values were based on an experiment which is measuring the difference between command velocities and measured velocities in NAVDATA. 

With respect to the Yaw, I calculated the angular velocity by differentiating the returned value of ARDrone::getYaw(). I wrote this kind of code and tested before.
```c++
#include "ardrone/ardrone.h"

int main(int argc, char *argv[])
{
    // AR.Drone class
    ARDrone ardrone;

    // Initialize
    if (!ardrone.open()) {
        std::cout << "Failed to initialize." << std::endl;
        return -1;
    }

    while (1) {
        // Key input
        int key = cv::waitKey(33);
        if (key == 0x1b) break;

        // Get an image
        cv::Mat image = ardrone.getImage();

        // Take off / Landing 
        if (key == ' ') {
            if (ardrone.onGround()) ardrone.takeoff();
            else                    ardrone.landing();
        }

        // Time [s]
        static int64 last = 0;
        double dt = (cv::getTickCount() - last) / cv::getTickFrequency();
        last = cv::getTickCount();

        // Orientation
        double yaw = ardrone.getYaw();
        std::cout << "deltaYaw   = " << yaw/dt << " [rad/s]" << std::endl;    // <- Yaw speed
 
        // Velocity
        double v_x, v_y, v_z;
        double velocity = ardrone.getVelocity(&v_x, &v_y, &v_z);
        //std::cout << "ardrone.vx = " << v_x << " [m/s]" << std::endl;
        //std::cout << "ardrone.vy = " << v_y << " [m/s]" << std::endl;
        //std::cout << "ardrone.vz = " << v_z << " [m/s]" << std::endl;

        // Move
        double vx = 0.0, vy = 0.0, vz = 0.0, vr = 0.0;
        if (key == 'i' || key == CV_VK_UP)    vx =  1.0;
        if (key == 'k' || key == CV_VK_DOWN)  vx = -1.0;
        if (key == 'u' || key == CV_VK_LEFT)  vr =  1.0;
        if (key == 'o' || key == CV_VK_RIGHT) vr = -1.0;
        if (key == 'j') vy =  1.0;
        if (key == 'l') vy = -1.0;
        if (key == 'q') vz =  1.0;
        if (key == 'a') vz = -1.0;
        ardrone.move3D(vx, vy, vz, vr);

        // Display the image
        cv::imshow("camera", image);
    }

    // See you
    ardrone.close();

    return 0;
}
```
However, the problem is that I cannot try the above code because I do not have AR.Drone on hand now.

puku0x

anfedres, 2015-06-20:
Thanks for your answer. I was wondering if you can tell me which are the values in the line of code below in order to leave vy, vx, vx and vr as percentages. Thank you. 

float v[4] = {-0.2f * (float)vy, -0.2f * (float)vx, 1.0f * (float)vz, -0.5f * (float)vr};

anfedres

anfedres, 2015-06-21:
puku0x?