puku0x/cvdrone #29: lopencv_contrib not found

alice60924, 2015-07-15:
Nice to see your blog!
I am a high-school student from Taiwan.
Also doing a cvdrone project recently, and found your open project inspiring.
When I managed to follow the instruction to install, I got the "lopencv_contrib not found" error.
I did install opencv both with macport and cmake; I've also add MODULE-PATH (opencv_contrib)to opencv by using cmake gui.
Thank you so much for reading my problem...
Any help will be appreciated!
Have a nice day!
Sincerely,
Alice

anfedres, 2015-07-20:
It seems more a OpenCV installation. First things first... 
1. Are using Ubuntu or Windows? 
2. What version of OpenCV did you use? 
3. If your are using Ubuntu, did you installed as a super user? 
4. Did you followed step by step the installation procedure? 

I'm sorry. but I'm not familiar with macport? Could you give me an insight?

alice60924, 2015-07-22:
Hi,
I am working on OS X,
Macport is just like Homebrew and does installation.
Macport automatically download the latest version of opencv and I do not
know if opencv_contrib is attached.
I had also used brew to download opencv3.0.
I did follow the tutorial step by step when installing cvdrone.
After I got the error of "missing opencv_contrib", I went for it and
considered it an add-module-path issue.

Thank you for helping!
Sincerely,
Alice.

puku0x, 2015-10-18:
Sorry for the inconvenience.

This is because the "opencv_contrib" module was separated from opencv modules since 3.0.
Makefile was fixed in https://github.com/puku0x/cvdrone/commit/594bf0f8937c324bc49a5a11140e95b14db0bd4d

Best regards,
puku0x

closed 2015-10-18 by puku0x