puku0x/cvdrone #30: Battery percentage and altitude percentage is always 0

valizada, 2015-10-14:
Hello,

Firstly thanks for the great library. I am having some problems with functions such as getBatteryPercentage() and getAltitude(). They always return 0(even though battery is fully charged).

Do you have any idea what could be a cause to this?

Thanks,
Fiz.

puku0x, 2015-10-15:
Hi

I had the same problem when run the sample program on Ubuntu (or OS X?).
What about switching **navdata_demo=TRUE** ?

```c
// ./src/ardrone/navdata.cpp
// AR.Drone 2.0
if (version.major == ARDRONE_VERSION_2) {
    // Disable BOOTSTRAP mode
    if (mutexCommand) pthread_mutex_lock(mutexCommand);
    sockCommand.sendf("AT*CONFIG_IDS=%d,\"%s\",\"%s\",\"%s\"\r", ++seq, ARDRONE_SESSION_ID, ARDRONE_PROFILE_ID, ARDRONE_APPLOCATION_ID);
    sockCommand.sendf("AT*CONFIG=%d,\"general:navdata_demo\",\"TRUE\"\r", ++seq); // HERE !!
    //sockCommand.sendf("AT*CONFIG=%d,\"general:navdata_demo\",\"FALSE\"\r", ++seq); // NOT USE
    if (mutexCommand) pthread_mutex_unlock(mutexCommand);
    msleep(100);

    // Seed ACK
    sockCommand.sendf("AT*CTRL=%d,0\r", ++seq);
}
```

If getBatteryPercentage() and getAltitude() work, it seems there is some bug in **ARDrone::getNavdata()**.

Another solution is to update the firmware of your drone to the latest version.

Thank you,
puku0x

valizada, 2015-10-15:
Hi, 

Thanks for the reply. I am running it on OS X. I have read about previous issues that had been opened, and have already changed navdata_demo to true, but it seems like  it doesn't help. In addition, the latest version of firmware is installed  on the drone. Probably there is another bug in getNavData function. 

Thanks, 
Fiz

puku0x, 2015-10-15:
Hi,

Thank you for your report.
I will check it tomorrow.

Regards,
pukuox

puku0x, 2015-10-18:
Hi,

Sorry for my late reply.
Unfortunately, I could not reproduce the issue.

--
AR.Drone Ver. 2.4.8
OS X 10.11 El Capitan
OpenCV 3.0.0.1 (via MacPorts)
CV Drone https://github.com/puku0x/cvdrone/commit/15dd04305d6b5f9d3a11c14666aa88fe831c207f

valizada, 2015-10-18:
Hi there,

Thanks for looking into this. 

I am not sure if this is what causes the issue, but I have OpenCV 2.4.12 (via brew).
I will upgrade, check again and report back.

Thanks,
Fiz.

valizada, 2015-10-25:
It worked after upgrading libraries. Thank you

closed 2015-10-25 by valizada

Chaaaarles95, 2018-04-06:
Please tell me how to upgrade the libraries on windows, i am having the same problem.