puku0x/cvdrone #31: using opencv 2.4.9

blueevangelion, 2015-11-15:
HI, thank you for you sources.
I am working with opencv now, but I don't like the opencv3.0 and some of the library I want to use is not included in your source. How can I Use OpenCV 2.4.9 with your sources?

Thank you

blueevangelion, 2015-11-16:
I want to use  "opencv2/xfeatures2d.hpp". But I cant find it in the library you created.

puku0x, 2015-11-21:
Hi,

Unfortunately, CV Drone does not provide opencv_contrib module.
You should build OpenCV 2.4.9 and opencv_contrib by yourself.

This is the last version which uses OpenCV 2.4.9. 
https://github.com/puku0x/cvdrone/tree/5e6ffc21e6e6b9b79d632e2991e1270e7f611f02
I'm not sure which environment are you using but this will help you how to set the path of libraries.

Thank you,
puku

closed 2016-06-04 by puku0x