puku0x/cvdrone #34: How to run a Windows example (2.0)

Zepolrotia92, 2016-01-15:
Good morning everyone!

Going back to this question (https://github.com/puku0x/cvdrone/issues/22), as I'm also newbie with this and I'm stuck in the same issue.

In my case (using W8/10) is it needed to modify the 'makefile' in (...)\cvdroneEDIT\build\mingw to run the samples? In that case, do I have to replace, for instance, the "PROGRAM = " statement (appearing 'Test.exe') just for 'kalman_tracking.cpp'?

This is what I've done until now, unsuccessfully.

Thank you for your kind support!

Zepolrotia92, 2016-01-15:
Do I perhaps need to substitute the content of 'main.cpp' file inside (...)\cvdroneEDIT\src for the one of the sample of my choice?

puku0x, 2016-01-24:
Perhaps yes.

Zepolrotia92, 2016-01-25:
Hi puku0x,

Thanks a lot for your help, I finally managed to take off my drone hahaha!

Could you please give me also a hand with this question (https://github.com/puku0x/cvdrone/issues/36), which is also mine -although it appears with a different nickname?

Thank you in advance!

Regards,
Aitor

closed 2016-02-11 by puku0x