puku0x/cvdrone #35: Not compatible with OpenCV 3.1

jplu, 2016-01-15:
Hello,

On Windows with Visual Studio 2012 and 2015, after changing the dependency from OpenCV 3.0 to OpenCV 3.1, the code crashs before to start with the following message:

```
The application was unable to start correctly (0x000007b).
Click on OK to close the application.
```

And the library ```opencv_hal``` doesn't exists anymore in OpenCV 3.1.

Any hint on how to update from OpenCV 3.0 to 3.1?

Thanks in advance.

puku0x, 2016-01-23:
Hi,

Thank you for your report.
I will try updating to OpenCV 3.1 soon !

Regard,
puku0x

jplu, 2016-01-24:
Hi,

After deeper investigate this issue, this error is not coming from OpenCV 3.1 as I succeed to make it works with.

The issue is coming from a switch from 32 bits to 64 bits. Have you try to compile your library in 64 bits?

Here what I did to make you able to reproduce the issue:

* Compile OpenCV 3.1 in 64 bits
* Compile FFMPEG in 64 bits
* Unzip the 64 bits library for pthread from the provided zip
* Open the Visual Studio solution
* Switch in the settings to 64 bits compilation
* Change the dependencie's name to link
* Compile and then execute

I did these steps with both Visual Studio 2012 and 2015 to get the above error.

Regards.

puku0x, 2016-01-24:
Hi,

Thank you for providing the details. 
I have never built my library in 64-bit environment but I will try it !

By the way, I updated the version of OpenCV which is bundled in CV Drone from 3.0 to 3.1 with opencv_contrib module. Please test it.

Regards,
puku0x

jplu, 2016-01-28:
It is working correctly, thanks! I close this issue and open a new one for a 64 bits version.

closed 2016-01-28 by jplu