puku0x/cvdrone #36: How to rotate the drone while maintaining it static

MisterTellini, 2016-01-20:
Hello everyone!

Thank you in advance for all the help provided not only through the construction of the repository, but also through your kind guidance in this forum.

I'm new in AR Drone programming (and **** in programming in general), and I just wanted to know whether it is possible to rotate the drone, for instance, 30 degrees by pressing one single key (I tried to type down higher numbers of vr than 1.0 without any success; I suppose it has to be related to the speed that the drone takes to accept this command ¿?).

Another question that comes to mind is simply how to manage to maintain the drone static in air after taking off. I have just changed all screws, but without any success also; the drone does whatever it wants and it takes a while to start obeying my instructions.

Thanks a lot, and have a nice day!

MisterTellini, 2016-01-20:
Stability is still a weak point in my code (please, help me out with that!), but another question is intriguing me:

I've already been able to make my drone rotate (easy peasy after all) after pressing a key, but I want to do it repeatedly. That is why I'm thinking of inserting my code inside an infinite loop. But my question is: how do I 'prioritize' some instructions to interrupt this rotation, given that one instruction cannot be carried out until the one in motion finishes? I would like, for instance, to move from some place to another (more and more questions coming from aside that I will try to face afterwards, hahaha) or to land de drone while it is rotating.

Thank you again, and see you soon!

MisterTellini, 2016-01-25:
Hello,

I had a go with my issue above, coding these following lines:

		// Rotation of the AR Drone
		double vx = 0.0, vy = 0.0, vz = 0.0, vr = 0.0;
		if (key == 'a'){

			if (key == 'f') break; // Stop rotating if 'f' button is pressed

			while (true) { // Infinite loop

				if (key == 'f') break; // Stop rotating if 'f' button is pressed

				for (int p = 0; p < 1500; p++){ // This value of p was settled 'randomly' -is there another option to select my own angle of rotation using degrees metrics?
					vr = 1.0;
					ardrone.move(vx, vy, vr);
					cv::imshow("camera", image);
				}
				Sleep(2000); // 2 seconds of break before starting moving again
			}
		} // Clockwise rotation

However I do find the following problems once I execute this:
- Drone's turnaround does not stop when I press 'f' button. It just stops once it finishes.
- Stabiization is also a great deal to take into account. I really need my drone to stay stopped while rotating, and it does not. Is there any way to perform this?
.. And last but not least, I'm looking for some way to control the drone through the sending of Wi-Fi waypoints. Does any of you know from where might I start to do so?

puku0x, 2016-01-26:
Wow, you are doing a loop in a loop.
Let's fix it at first.