puku0x/cvdrone #37: multicolor tracking

rezaprasetyo, 2016-01-26:
hello there, you help me alot on my final project.

can you help me about tracking multicolor? because i need to track like green,blue,red, and black in ar drone camera.

thanks :)

puku0x, 2016-01-26:
This is it !

```c++
#include "ardrone/ardrone.h"

// Thresholds
class Thresholds {
public:
    Thresholds() {
        minH = minS = minV = 0;
        maxH = maxS = maxV = 255;
    }
    int minH, maxH;
    int minS, maxS;
    int minV, maxV;
};

// Find the largest object in a binalized image
cv::Rect findLargestBoundingRect(cv::Mat img)
{
    cv::Rect rect;
    cv::Mat tmp = img.clone();
    
    // De-noising
    cv::Mat kernel = getStructuringElement(cv::MORPH_RECT, cv::Size(3, 3));
    cv::morphologyEx(tmp, tmp, cv::MORPH_CLOSE, kernel);
    
    // Detect contours
    std::vector< std::vector<cv::Point> > contours;
    cv::findContours(tmp, contoursRed, cv::RETR_CCOMP, cv::CHAIN_APPROX_SIMPLE);
    
    // Find largest contour
    int contour_index = -1;
    double max_area = 0.0;
    for (size_t i = 0; i < contours.size(); i++) {
        double area = fabs(cv::contourArea(contours[i]));
        if (area > max_area) {
            contour_index = i;
            max_area = area;
        }
    }
    
    // Object detected
    if (contour_index >= 0) {
        rect = cv::boundingRect(contours[contour_index]);
    }
    
    return rect;
}

// --------------------------------------------------------------------------
// main(Number of arguments, Argument values)
// Description  : This is the entry point of the program.
// Return value : SUCCESS:0  ERROR:-1
// --------------------------------------------------------------------------
int main(int argc, char *argv[])
{
    // AR.Drone class
    ARDrone ardrone;
    
    // Initialize
    if (!ardrone.open()) {
        std::cout << "Failed to initialize." << std::endl;
        return -1;
    }
    
    // XML save data
    std::string filename("thresholds.xml");
    cv::FileStorage fs(filename, cv::FileStorage::READ);
    
[... 109 lines cut ...]
        cv::write(fs, "S_MAX_GREEN", thGreen.maxS);
        cv::write(fs, "S_MIN_GREEN", thGreen.minS);
        cv::write(fs, "V_MAX_GREEN", thGreen.maxV);
        cv::write(fs, "V_MIN_GREEN", thGreen.minV);
        cv::write(fs, "H_MAX_BLUE",  thBlue.maxH);
        cv::write(fs, "H_MIN_BLUE",  thBlue.minH);
        cv::write(fs, "S_MAX_BLUE",  thBlue.maxS);
        cv::write(fs, "S_MIN_BLUE",  thBlue.minS);
        cv::write(fs, "V_MAX_BLUE",  thBlue.maxV);
        cv::write(fs, "V_MIN_BLUE",  thBlue.minV);
        cv::write(fs, "H_MAX_BLACK", thBlack.maxH);
        cv::write(fs, "H_MIN_BLACK", thBlack.minH);
        cv::write(fs, "S_MAX_BLACK", thBlack.maxS);
        cv::write(fs, "S_MIN_BLACK", thBlack.minS);
        cv::write(fs, "V_MAX_BLACK", thBlack.maxV);
        cv::write(fs, "V_MIN_BLACK", thBlack.minV);
        fs.release();
    }
    
    // See you
    ardrone.close();
    
    return 0;
}
```

rezaprasetyo, 2016-02-02:
thank you very much for the program, and now, can you tell me how to get
the sensor data from the quadrotor?

thanks!

puku0x, 2016-02-02:
Here is a sample code which gets sensor values from AR.Drone. 
https://github.com/puku0x/cvdrone/blob/master/samples/sample_navdata.cpp

More API references are in https://github.com/puku0x/cvdrone/wiki/API-reference

Regards,
puku0x

rezaprasetyo, 2016-02-02:
anyway, the multi color tracking isnt working. help please

rezaprasetyo, 2016-02-02:
visual studio said "thresholds is undefined" help me thanks

puku0x, 2016-02-02:
Did you add Thresholds class ?
```c++
// Thresholds
class Thresholds {
public:
    Thresholds() {
        minH = minS = minV = 0;
        maxH = maxS = maxV = 255;
    }
    int minH, maxH;
    int minS, maxS;
    int minV, maxV;
};
```

abraxas4, 2016-07-18:
hello guys,
I'm new here, but I was trying the code puku0x wrote above (Thank you very much)
and it looks like the argument order is opposite with cv::Scalar(...).
```
        cv::rectangle(image, findLargestBoundingRect(red),   cv::Scalar(**255**,0,0));
        cv::rectangle(image, findLargestBoundingRect(green), cv::Scalar(0,255,0));
        cv::rectangle(image, findLargestBoundingRect(blue),  cv::Scalar(0,0,**255**));
```
->
```
        cv::rectangle(image, findLargestBoundingRect(red),   cv::Scalar(0,0,**255**));
        cv::rectangle(image, findLargestBoundingRect(green), cv::Scalar(0,255,0));
        cv::rectangle(image, findLargestBoundingRect(blue),  cv::Scalar(**255**,0,0));
```
If this is not right, just ignore it.
:)

nnguyen16, 2017-03-02:
hello guys,
I use the same code and it said "'contoursRed': undeclared identifier" 
how do i fix this?

Thanks