puku0x/cvdrone #38: Detect tags

venkathindusthani, 2016-01-26:
Hi,
Your cvdrone application is very useful to me in my project. I use your application code for reference. I liked the design how you developed. 
I'm thinking to integrate April Tags detection libraries with this application. Can you help me on this, if you have already done similar kind of integration with your application?
Thank you so much in advance,

puku0x, 2016-01-28:
Hi,

Thank you for your comment.

I wrote a maker-based AR sample before.
https://github.com/puku0x/cvdrone/blob/master/samples/sample_marker_based_ar.cpp

The sample uses this kind of marker, similar to April Tags..
[image]

puku0x, 2016-01-28:
One more thing !
Recently, CV Drone bundled OpeCV3.1 with [extra modules](https://github.com/Itseez/opencv_contrib) .

You can also use the OpenCV's ArUco integration.
http://docs.opencv.org/master/d5/dae/tutorial_aruco_detection.html#gsc.tab=0
https://github.com/Itseez/opencv_contrib/blob/master/modules/aruco/samples/detect_markers.cpp

venkathindusthani, 2016-01-29:
Thank you so much,
I will explore into the topics u mentioned

closed 2016-04-30 by puku0x