puku0x/cvdrone #39: Crash in 64 bits version

jplu, 2016-01-28:
Hello,

An issue is coming from a switch from 32 bits to 64 bits.
Here what I did to make you able to reproduce the issue:

* Compile OpenCV 3.1 in 64 bits
* Compile FFMPEG in 64 bits
* Unzip the 64 bits library for pthread from the provided zip
* Open the Visual Studio solution
* Switch in the settings to 64 bits compilation
* Change the dependencie's name to link
* Compile and then execute

I did these steps with both Visual Studio 2012 and 2015 to get the below error:

```
The application was unable to start correctly (0x000007b).
Click on OK to close the application.
```

Regards.

puku0x, 2016-01-28:
Thank you very much !
I will prepare for it.

puku0x, 2016-01-28:
Hmm,
http://answers.microsoft.com/en-us/windows/forum/windows_7-performance/error-the-application-was-unable-to-start/05a2b904-3f61-4d08-94d6-e2ff92161111?auth=1

jplu, 2016-01-28:
Yep, I came up to that thread as well, but I checked what they say and all my DLLs for OpenCV, FFMPEG and Pthread are in 64 bits :(

JAbdor7, 2016-11-02:
Good day,

I was wondering if you were able to compile the cvdrone library for x64 in visual studio?

jplu, 2016-11-03:
No I gave up :(

JAbdor7, 2017-05-01:
I think I found the solution. There seems to be an issue with the Linker Manifest File. After I turn that off I don't get the error. Now since I don't have the drone with me I can't test it, however I do get an error that it can't connect to the drone. I'll post if I am able to get it working in two days once i have access to the drone.

Youssef-Salama, 2017-05-02:
@JAbdor7 I have the drone with me. If you could forward me the code along with instructions on how to run it, I will test it and let you know. I need this urgently for a college research. Thank you.

JAbdor7, 2017-05-02:
I was able to run cvdrone on x64 and it was tested with the drone.