puku0x/cvdrone #41: ardrone.move3D(vx, vy, vz, vr):  Doesn't go forward when giving valid vx value!

lhg912, 2016-01-31:
Hi, thanks for giving your great code ^^.

But I have a problem. Although I gave valid vx or vy value, my drone hardly goes forward or backward,
left or right. My drone rotates well(vr) and changes its altitude well(vz), but it doesn't go in parallel directions well, when using ardrone.move3D(vx, vy, vz, vr), with valid vx, vy values. 
It seems that it slightly moves, but pretty hard to notice the movement.

puku0x, 2016-02-01:
Thank you for your report.

Is that happen when just sending move3D(1.0, 0.0, 0.0, 0.0) ?
I mean doing this.
```C++
// Main loop
while (1) {
    // Key input
    int key = cv::waitKey(33);
    if (key == 0x1b) break;

    // Take off / Landing 
    if (key == ' ') {
        if (ardrone.onGround()) ardrone.takeoff();
        else                    ardrone.landing();
     }

    // Just sending vx=1.0, not using key inputs !
    double vx = 1.0, vy = 0.0, vz = 0.0, vr = 0.0;
    ardrone.move3D(vx, vy, vz, vr);
```

I heard this kind of problem often happens due to the keyboard repeat.

Hope it helps,
puku0x

nicklas669, 2016-06-03:
> my drone hardly goes forward or backward,
left or right. My drone rotates well(vr) and changes its altitude well(vz), but it doesn't go in parallel directions well, when using ardrone.move3D(vx, vy, vz, vr), with valid vx, vy values. 

I am experiencing this exact problem aswell. Has any fix been found? I tried ardrone.move3D(1.0, 0.0, 0.0, 0.0) and the drone moved forward.

cnpcshangbo, 2017-03-13:
I encountered the same problem. I found that the letter keys and the arrow keys have different repeating behavior. Therefore, I guess changing all of the keys to use letters can solve this problem.

mikerosz, 2017-10-11:
Hello everyone, I like this program sample_tracking. I am trying to use this code to tell the drone to follow a straight line when a specific object is detected. IN MY CASE, A GUTTER, once the color white is detected, tell the drone to go from one end to the other. any way I can do this maybe drawing a line within the box and somehow telling drone to follow line. Any help would be appreciated.