puku0x/cvdrone #42: Autonomous flight

Zepolrotia92, 2016-02-01:
Hi everyone,

Apart from my [previous problem](https://github.com/puku0x/cvdrone/issues/36), which I have not been able to resolve yet, I was wondering if there is any tool in this SDK to make the drone fly 'autonomously', this is, through the use of waypoints.

Thank you in advance, and have a nice day!

puku0x, 2016-02-01:
Unfortunately, CV Drone only provides
- Getting camera image
- Getting sensor data (gyro, velocity, altitude, ...)
- Sending commands
- Many available OpenCV functions

You should make it with your own hands.

Thanks,
puku0x

mikerosz, 2017-10-11:
Hello everyone, I like this program sample_tracking. I am trying to use this code to tell the drone to follow a straight line when a specific object is detected. IN MY CASE, A GUTTER, once the color white is detected, tell the drone to go from one end to the other. any way I can do this maybe drawing a line within the box and somehow telling drone to follow line. Any help would be appreciated.