puku0x/cvdrone #43: Help with make error

vrobot, 2016-02-07:
Hello,

I am a novice and need help with an error I get when compiling cvdrone in lubuntu 14.0.3LTS.
The OS came with opencv 2.4.8 but I downloaded opencv 3.1 and compiled and install it.
Then when I compiled cvdrone it failed in last step with library opencv_hal not found.
I read that opencv_hal was depricated in opencv 3.1 and so uninstalled it
and compiled opencv 3.0 and installed it.
Now when I compile cvdrone with just "make" without any change to the unix makefile I get the error:
...
...
g++ -O2 -Wall -D__STDC_CONSTANT_MACROS    -c -o ../../src/main.o ../../src/main.cpp
g++ ../../src/ardrone/ardrone.o ../../src/ardrone/command.o ../../src/ardrone/config.o ../../src/ardrone/udp.o ../../src/ardrone/tcp.o ../../src/ardrone/navdata.o ../../src/ardrone/version.o ../../src/ardrone/video.o ../../src/main.o -o test.a -O2 -Wall -D__STDC_CONSTANT_MACROS   -lm -lpthread -lavutil -lavformat -lavcodec -lswscale -lopencv_calib3d -lopencv_core -lopencv_features2d -lopencv_flann -lopencv_hal -lopencv_highgui -lopencv_imgcodecs -lopencv_imgproc -lopencv_ml -lopencv_objdetect -lopencv_photo -lopencv_stitching -lopencv_superres -lopencv_video -lopencv_videoio -lopencv_videostab 
../../src/ardrone/video.o: In function `ARDrone::operator>>(cv::Mat&)':
video.cpp:(.text+0x1ce7): undefined reference to `cv::cvarrToMat(void const*, bool, bool, int, cv::AutoBuffer<double, 136u>*)'
../../src/main.o: 
In function `main':
main.cpp:(.text.startup+0x382): undefined reference to `cv::cvarrToMat(void const*, bool, bool, int, cv::AutoBuffer<double, 136u>*)'
main.cpp:(.text.startup+0x482): undefined reference to `cv::String::allocate(unsigned int)'
main.cpp:(.text.startup+0x4a3): undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
main.cpp:(.text.startup+0x4ab): undefined reference to `cv::String::deallocate()'
main.cpp:(.text.startup+0x6c4): undefined reference to `cv::String::deallocate()'
collect2: error: ld returned 1 exit status
make: *** [test.a] Error 1

Any ideas how to fix this ?  Thanks a lot !

puku0x, 2016-02-11:
Hi,

Thank you for your report.

Unfortunately, I have never installed OpenCV 3.x on Ubuntu and I have no idea to solve the problem now.

Thanks,
puku0x

Yevgeniyr1, 2016-12-23:
Any updates on this? I ran into exactly the same error.

puku0x, 2017-02-11:
@Yevgeniyr1 
Thank you for the information about `CXXFLAGS` !
It is applied to the current makefile.