puku0x/cvdrone #44: ATAM integration

puku0x, 2016-02-11:
Hi all,

Thank you for using CV Drone !
This time, I would like you to help me.

Do you interested in SLAM ?
I'm fed up with making a sample which just detects a color marker and tracks it.

ATAM seems to be easy to integrate.
https://github.com/CVfAR/ATAM

It is high time to CV Drone to be high-leveled library, isn't it ?

Best regards,
puku0x

puku0x, 2016-02-22:
No, abort it.

ATAM uses "cvsba" which has **GPL3** license.

To avoid the GPL infection, we must

- Replace the cvsba to OpenCV's BA function (cv::LevMarqSparse ?)

or

- Add a batch file to download and install the cvsba library

Hmm, this needs more time to integrate.

puku0x, 2016-02-28:
It's better to start from a simple implementation.
I found this, the monocular visual odometory using cv::findEssentialMat.

https://github.com/avisingh599/mono-vo

Can anyone help me ?

valizada, 2016-03-01:
I would be happy to start helping after 18th of march. Could you try to set some tasks to be done?

puku0x, 2016-03-01:
Wow, that's great !

The aim of this story is to implement an autonomous flight function to CV Drone.
 
Now I'm planning to
・Make a Visual-Odometory sample using [mono-vo](https://github.com/avisingh599/mono-vo)
・Extend it to Visual-SLAM (without Bundle adjustment) referring to [ATAM](https://github.com/CVfAR/ATAM)
・Add Bundle adjustment
・Add Extended Kalman Filter (optional)
then, 
・Make a point-to-point moving sample

closed 2016-06-04 by puku0x