puku0x/cvdrone #47: What happens when no commands sent to the drone?

valizada, 2016-02-28:
I wonder what commands do you send to the drone to stay in the air when no user input is sent. 

I have noticed that sometimes when drone starts flying(after takeoff command) it would simply rotate on its place, which does not always happen. 

I am mainly running the drone in a loop which simply reads images (getImage command) and occasionally I send some move3d commands. And I do want the drone to simply hover when there are no commands sent to it. However, I am struggling to understand how. 

Beside that, if for a while no commands sent it will just go up to the ceiling.

Simply said, sometimes the drone performs some random manoeuvres. Would be nice if you could say your opinion about it.

I saw another thread as well asking "how to make it hover", which had no responses. I understand that it could be something difficult to achieve and that is why I am not asking for an explicit way of solving but just your opinions. 

Thanks,
Fiz

valizada, 2016-02-29:
With a reference of a friend, I found that there is a control:flying_mode command which changes between FreeFlight mode and HOVER_ON_TOP_OF_ROUNDEL. As I noticed, you did not set this control unit in ARDrone::initCommand and I am not sure which one it uses by default. So, I added following:
`        // flying_mode is set to FreeFlight mode
        sockCommand.sendf("AT*CONFIG_IDS=%d,\"%s\",\"%s\",\"%s\"\r", ++seq, ARDRONE_SESSION_ID, ARDRONE_PROFILE_ID, ARDRONE_APPLOCATION_ID);
        sockCommand.sendf("AT*CONFIG=%d,\"control:flying_mode\",\"%d\"\r", ++seq, 0);
        msleep(100);`

I will test it tomorrow and let u know if it solved the problem.

puku0x, 2016-03-01:
Hi,

Thank you for your report !
I'm looking forward to the result.

Regards,
puku0x

valizada, 2016-03-01:
Hi there,

It did not work. Still happens. I noticed that it happens (rotating after starting) after I crashed it. So, I suspect it can be because of some kinda calibration it is doing by itself. However, I cannot recall if it happens always after the crashing. 

In general, if you have any idea about calibrating it, I mean when I should do which calibration? In general, what they do?

Besides, in command.cpp file I tried to set max altitude, but sometimes it still flies too high. I have no idea why. When we set the max altitude doesn't it mean that the drone won't fly any higher even if we explicitly send commands to it to do so?

Thanks,
Fiz