puku0x/cvdrone #56: CUDA Support

JAbdor7, 2016-10-20:
Good day,

Is it possible to add CUDA support for Opencv?

Thanks

puku0x, 2016-10-31:
Hi,

Yes, you can add the CUDA support by building OpenCV library with WITH_CUDA option.

You need CUDA SDK and OpenCV, CMake, Visual Studio for Windows.
For UNIX, please refer to http://www.pyimagesearch.com/2016/07/11/compiling-opencv-with-cuda-support/

Unfortunately, I've never used CUDA. The official tutorial may help you.
http://docs.opencv.org/3.1.0/d3/d52/tutorial_windows_install.html

Regards

JAbdor7, 2016-11-01:
Thanks for the information, I've tried without successs with visual studio but it won't work. Does the CvDrone only work with OpenCV 3.1?

puku0x, 2017-02-11:
@JAbdor7 
CV Drone works with either OpenCV 2.4.x and 3.x. 
(Current version bundles only 3.1 with opencv_contrib)

Youssef-Salama, 2017-04-09:
Is there a 64bit version of CV drone. My entire code is in 64 bit and it uses CUDA 8 supported opencv libraries. Unfortunately, CUDA 8 doesn't have a 32bit version. What should I do?

JAbdor7, 2017-04-24:
Are you able to release a 64 bit version of cv_Drone

JAbdor7, 2017-04-24:
Are you able to release a 64 bit version of cv_drone?

puku0x, 2017-04-24:
@JAbdor7 Unfortunately, I have no time to update CV Drone now because I should finish two apps until June and July, I'm really sorry.

Youssef-Salama, 2017-04-29:
You r library was so nice, it's a shame. My issues is that I am using CUDA
which doesn't support 32 bit anymore and I need to somehow control the
drone on a 64 bit platform. Can you help with that at all?
I need this within 3 weeks for my final part of my Honours Thesis at
Western Sydney University. I appreciate your time.

JAbdor7, 2017-04-29:
Compile opencv in 64 bit, switch pthreads and ffmpeg librarys to 64 bit and build.

closed 2017-05-02 by JAbdor7